meshtastic/device-ui #3: Help a Newbie Out: Add Usage Instructions to README

Mearman, 2024-05-01:
Hey there! I'm excited to dive into the GUI repo, even though it's not officially launched yet.
I know the GUI isn't "camera ready" yet, but I'm eager to test and contribute.
Could we get some clear usage instructions added to the README file?
It'd be super helpful to have step-by-step guidance on building and installing the package.

mverch67, 2024-05-01:
This repo is under development and doesn't even compile right now due to ongoing tool chain changes and refactoring activities. When it's ready the firmware will be available at the usual flasher site. Please be patient for a few more months.

Reid-n0rc, 2024-06-17:
@mverch67 Several people, including myself are interested in contributing to the project. Would you be willing to share the current tool chain so we can help?

mverch67, 2024-06-21:
This repository is a library which is integrated into meshtastic firmware. It cannot be build by its own without a project. The tool chain is the same as used by meshtastic, there is not much difference. Refer to meshtastic documentation.

closed 2024-06-21 by mverch67