meshtastic/device-ui #9: Support Emojis

mattdrum, 2024-06-25:
Not sure if planned or not but it would be helpful if you could support emjois in the UI for Short Name, Long Name, and messages. Several nodes around here have been using emojis as a convention for device name grouping and it is very confusing on the current Tdeck display.

mverch67, 2024-06-26:
I know there were also requests for non-latin fonts. As the device cannot store a number of different fonts and/or glyphs for all types of desires (and the device-ui library is not a place to copy all possible fonts into) my idea is to load a font stored on SD card. So if somebody wants a certain font with emojis he could load that font, another one who likes Japanese could store and load such font, and another one who likes arabic could search for that, download to SD card and use that.

mattdrum, 2024-06-26:
I am not an expert on fonts but I was under the impression that Unicode is how the meshtastic is sending the emoji payload. Unicode contain definitions for all characters, including non-latin. On my android meshtastic app, the unicode is rendered based on it's own unicode library. I know there is some variation across platforms. Is this the font you are referring to?

mverch67, 2024-06-26:
Yeah, the emojis are part of a font which defines a set of characters and these must be translated into lvgl compatible bitmaps. The unicode is just an index into the font library to pick a certain character to be displayed.

As an example you can look at a bitmap how a latin font is encoded:
https://github.com/meshtastic/device-ui/blob/0b5643f659e77f3ddbcd1f602f62d12448f817d9/generated/ui_320x240/ui_font_montserrat_16.c#L19

mattdrum, 2024-06-26:
Ah, I see. Thanks for the clarification. I'll leave you guys alone as you work on this. I can't wait to see the end results!

closed 2024-06-26 by mattdrum