meshtastic/device-ui #24: T-Deck Critical fault #11 when typing long single word message

MonstaB, 2024-12-16:
### Category

Other

### Hardware

T-Deck

### Firmware Version

latest bata

### Description

can trigger this at will by either sending a pvt message that is max character length or by typing characters in a single word too long to fit on the screen without a space eg.. repeating letter KKKKKKKKKKKK all the way accross.
tell me how to get a log and ill do it

joyel24, 2024-12-16:
@MonstaB I just discovered you duplicate this issue here... So I assume it's concerning the TFT-UI and not the public firmware?
Can you answer plz? https://github.com/meshtastic/firmware/issues/4275#issuecomment-2545868067

mverch67, 2024-12-16:
This old issue has been fixed / is not reproducible anymore so I'll close it.

[image]

closed 2024-12-16 by mverch67

joyel24, 2024-12-16:
@mverch67 Perfect 👍

By trying to reproduce this one I got another bug here : https://github.com/meshtastic/firmware/issues/4275#issuecomment-2545771251

Sorry, i was trying to clarify this issue that was lacking a lot of Informations.
I own T-deck for only 3 days and don't know the whole issues/fixes history yet...