meshtastic/device-ui #45: Region Unset, when its actually set

DomTheDorito, 2025-02-26:
On the homepage on the T-Deck, it says region unset, however the region is indeed set in the settings. The radio and messages function as normal.

mverch67, 2025-02-26:
As a workaround you can long press 2x the radio tower symbol next to the text region unset. If it then has retrieved the region information it should also display the frequency correctly.

mverch67, 2025-02-26:
I saw it's more or less a one-time issue... maybe the firmware does not send the right information at the first start. Had no chance to debug it yet as it is hardly reproducible.

mverch67, 2025-02-26:
> It did reboot after that, and maintain frequency after that; unsure if that's normal behavior or not.
Usually it does not reboot when changing tx_power. If it does I would say it's also an indication that some state was not correct in the firmware.

jpryor223, 2025-03-01:
Had the same experience, I had initially set the Lora info from the serial client app versus the UI.  The workaround solved this for the T-deck.

aveao, 2025-08-26:
This happens, AFAICT (at least on TFTView_320x240), when MUI gets a channel number of 0 (it's supposed to be 1 or more, 0 signifies "calculate locally from channel hash").

The code calculating frequency uses the channel, `channel - 1` rather, and as channel is a uint32, it overflows, calculating frequency numbers to the likes of 1073742720 MHz.

The code decides to not display frequency values smaller than 1 or larger than 100000 MHz, and in all cases where it does that, it says "region unset". Annoyingly/amusingly this also is not a surefire way to detect if region is unset, I think this could still display a number when region is set.

There's code that calculates the channel number from the channel hash, but that's called after the lora frequency is displayed when MUI is receiving the config. It's also using a different reference to the channel number (`db.config.lora.channel_num` vs `cfg.channel_num`, only former is recalculated, latter is a copy of what's received from meshtastic).

The reason why turning tx off and on works is because it uses `db.config.lora` to call `showLoRaFrequency` (and maybe because changing the config hardcodes the channel number that meshtastic sends in the future).

I've sent a PR that (#186):
- Properly checks if region is unset
- Moves the channel calculation earlier
- Makes the lora frequency display use the `db.config.lora.channel_num` when calculating frequency instead.

closed 2025-08-27 by mverch67