meshtastic/device-ui #51: bug: Boot screen gets clicked accidentally, restarts the device to Programming mode

iliaschat, 2025-02-27:
**Describe the bug**
The T-Deck Plus, when using the original case, has a reset button on the side that can be easily pressed unintentionally (especially if it's in a pocket). If an unexpected reboot occurs and the screen's sensor detects a touch, it will immediately restart in programming mode.

**To Reproduce**
Steps to reproduce the behavior:
1. While on the boot screen, touch the meshtastic logo once.
2. The device reboots into 'programming mode.'

**Expected behavior**
While on the boot screen, touch and hold the meshtastic logo until the 'loader' animation is completed, then reboot.

**Device (please complete the following information):**
 - Type: T-Deck Plus
 - Exact version 2.6.0@19951e3b6a457b577a9d43a290300d11c52bfe9c]

mverch67, 2025-02-27:
Best solution:
https://www.ebay.com/itm/316130177061 or 3D print your own.

joyel24, 2025-02-27:
It's better like this to me, cause when programming it, it reboots very often so better to come back alone to Programming mode.

closed 2025-02-27 by mverch67

iliaschat, 2025-02-27:
> It's better like this to me, cause when programming it, it reboots very often so better to come back alone to Programming mode. You can add something next to your button to avoid accidental pushes on it..

I totally agree with that. This is not what I meant. 
The problem is when restarting in normal mode and it happens to accidentally touch the display..
Then it performs an unwanted action. 

> Best solution: https://www.ebay.com/itm/316130177061 or 3D print your own.
> 
> Or print this from AlleyCat:
> 
> [image]

Thanks for the suggestion. I'll give the 3D print a try, nevertheless