meshtastic/device-ui #62: feat: T-Deck w/ 2.6 MUI, Maps does not show fixed position when GPS is unavailable

antijingoist, 2025-03-01:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.6.0.0b106d4

### Description

On T-Decks without a GPS, GPS set to NOT_PRESENT, and using fixed_position to manually set location:

Opening the Maps tab on the T-Deck defaults to Big Ben in UK, and assumes device position is at that location. 

Device position is still sent out accurately, this only appears to affect the maps.

Expected: 
Location set using fixed position would be what Maps shows.

### Relevant log output

```Shell

```

closed 2025-03-02 by mverch67

rfpmachado, 2025-06-26:
How do I  revert the "persistent home position which can be set by long pressing the crosshair button"? Now my T-Deck reports the correct position to the network, but the marker on the map doesn't change.
Thanks.

mverch67, 2025-06-27:
If you have GPS enabled then there are two markers: a current GPS position (which may move if the node reports a new position) and a fixed home position which you can set manually.

dzienisz, 2025-08-28:
I still don't get it I am on 2.7.5 and I can't make my build in GPT in T Deck Plus

PR #65 (2025-03-02): feat: set fixed_position using MUI map home position
set fixed_position lat/lon (if enabled) during setting home position if this switch has been enabled
fixed position is displayed as greyed out to distinguish from actual GPS readings which are colored

closes #62