meshtastic/device-ui #68: Seeed Studio SenseCAP Indicator is excessively rebooting

jpryor223, 2025-03-03:
**Describe the bug**
The Seeed Studio SenseCAP Indicator is excessively rebooting with Preview 2.6.0 and Meshtastic device UI.  For example I have a serial log file showing 18 reboots in a ~30 minute timeframe.  At best I have seen the device not reboot for about ~2 hours.  The logs indicate unhandled exceptions, and at times another reboot occurs while the device is starting back up.

I have interacted with the device via the UI in different ways and continue to see reboots from leaving the device on various screens.  I have continued to see reboots with CLIENT, CLIENT MUTE and CLIENT HIDDEN roles.  I have used a few different USB cables and power supplies and continue to see the same reboot issues.

**To Reproduce**
Steps to reproduce the behavior:
1. Flashed the SenseCAP Indicator with Preview 2.6.0 and Meshtastic device UI
2. Turned on the device and setup with US and mostly defaults
3. Wait a bit and watch the device continue to Rebooting...
4. View unhandled exception errors in serial debug logs

**Expected behavior**
The SenseCAP Indicator device does not excessively reboot for unhandled exceptions.  For comparison running Preview 2.6.0 with MUI on a T-Deck and has gone at least 48 hours without unexpected reboot.  The same SenseCAP Indicator on stable Beta 2.5.20 firmware did not have rebooting issues.

**Screenshots / Photos**
Screen shows "Rebooting..." and then the device restarts...

**Device (please complete the following information):**
 - Type: Seeed Studio SenseCAP Indicator Model D1L
 - Exact version: 2.6.0.0b106d4 Tech Preview
 - Firmware download source: web-flasher
 - Other attached HW: N/A

**Additional logs**
Logs from ~30 minute timeframe has 18 reboots:
16 reboots after the following error:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
2 reboots shortly after rebooting with:
Stack smashing protect failure!
The events preceding  the unhandled exceptions and rebooting events were various scenarios based on the logging, not seeing one consistent type of activity leading up to the issue.

[meshtastic-log-2025-03-01T17-03-38.962Z.log](https://github.com/user-attachments/files/19047255/meshtastic-log-2025-03-01T17-03-38.962Z.log)

mverch67, 2025-03-03:
Unfortunately, there are no .elf files provided with this preview that are needed for decoding the stack dump.

mverch67, 2025-03-06:
It seems your observation is not easily reproducible. Can you please do a factory reset and try again with default values (except region) and check if you still observe the issue. Then, as a second step configure the device with your further config until it goes into the boot loop again. Finally, describe the exact settings you made.

thebiggestgit, 2025-03-19:
I too have the sensecap indicator and getting reboots on the latest 2.6.1 as well as on the technical preview version. In EU_868 region. Seems better if I leave the device alone, but still will reboot - as I have to touch the top button to wake it and use.

thebiggestgit, 2025-03-21:
I have tried the latest alpha release 2.6.2 (with full erase) and still exibits rebooting when i have a second channel added (which I did by scanning QR code on android app), no config restore was carried out as i wondered if that might have been the issue.  That's when just leaving alone and not doing anything apart from waking it up with green button  When I removed my private channel (which does work on other devices such as my t-deck plus with previous firmware 2.6.1 and didn't  reboot like this)  so far it appears to be "stable"... though haven't used in anger yet.... I wonder if that was the issue, did you have more than longfast enabled?

thebiggestgit, 2025-03-21:
I've tried the latest alpha 2.6.2 release and still doing the same, although seems to reboot less when I only have the long fast channel and not my own private one added.

thebiggestgit, 2025-03-21:
> Are you able to share logs and your exported configuration?

Will see if I can..

jpryor223, 2025-03-22:
Continued to see now with latest 2.6.1 and 2.6.2, I had tried fresh installs.  At times I can see it start to reboot before even changing any settings while still in Unset mode.  Have been trying to change bare minimum settings.  At most I have see the unit go 8 hours during the day or about 20 hours over night.  But then we reboots start occurring, usually after touch or other interaction, not finding a great way to fix and just turn it back off for a while.  I also ran for a few days using 2.6.0 without MUI and did not have any issue when just running the preview firmware.

I do see the ELF file mentioned in the log now, please help confirm what steps I need to take, is there added file I need to be able to extract?  Was seeing an ELF mentioned here: https://meshtastic.org/docs/development/firmware/stacktraces/

Example log:

[0mGuru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42051635  PS      : 0x00060530  A0      : 0x820330d5  A1      : 0x3fcb14f0  
A2      : 0x3fcb152c  A3      : 0x3fcb158c  A4      : 0xffffffff  A5      : 0xffffffff  
A6      : 0xffffffff  A7      : 0x000000ff  A8      : 0x00000040  A9      : 0x3fcb14d0  
A10     : 0x00000080  A11     : 0x00000080  A12     : 0x00000003  A13     : 0x3fcb152c  
A14     : 0x00000180  A15     : 0x00000000  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000b  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x42051632:0x3fcb14f0 0x420330d2:0x3fcb1570 0x4202dec6:0xffffffff |<-CORRUPTED

ELF file SHA256: d2044debfc270ab7

Rebooting...

mverch67, 2025-03-22:
```
$HOME/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line -e firmware-seeed-sensecap-indicator-tft-2.6.2.31c0e8f.elf -pfiaC 0x42051632:0x3fcb14f0 0x420330d2:0x3fcb1570 0x4202dec6:0xffffffff 
0x42051632: LGFXDriver<LGFX_INDICATOR>::touchpad_read(_lv_indev_t*, lv_indev_data_t*) at :?
0x420330d2: lv_indev_read_timer_cb at :?
0x4202dec6: lv_timer_periodic_handler at ??:?
```

jpryor223, 2025-03-23:
I have 8 of the above identical backtraces across an 8MB log file which was about 12 hours. 

Then I have two other reboots in that log, these two additional were reboots occurring just after a previous reboot:

Stack smashing protect failure!

Backtrace: 0x40377d92:0x3fcb1490 0x403816fd:0x3fcb14b0 0x403773da:0x3fcb14d0 0x4205167d:0x3fcb14f0 0x420330d2:0x3fcb1570 0x4202dec6:0x40404040 |<-CORRUPTED

ELF file SHA256: d2044debfc270ab7

Rebooting...

Advise if needing more from the log or any other details.

mverch67, 2025-03-23:
```
0x40377d92: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
0x403816fd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
0x403773da: __stack_chk_fail at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/stack_check.c:35
0x4205167d: LGFXDriver<LGFX_INDICATOR>::touchpad_read(_lv_indev_t*, lv_indev_data_t*) at :?
0x420330d2: lv_indev_read_timer_cb at :?
0x4202dec6: lv_timer_periodic_handler at ??:?
```

mverch67, 2025-03-23:
@jpryor223 
I've increased the stack size. Maybe you can try this new build if it still has these issues:

[firmware-seed-sensecap-indicator-tft-2.6.2.1ee800e90-update.zip](https://github.com/user-attachments/files/19411430/firmware-seed-sensecap-indicator-tft-2.6.2.1ee800e90-update.zip)

mverch67, 2025-03-26:
> I have three Indicators all with the reboot issue. All running 2.62 now. They also misbehave on 2.60 / 2.61

Can you try the update .bin above (or wait for 2.6.3). I've increased the stack size.

resunltd, 2025-03-26:
> > I have three Indicators all with the reboot issue. All running 2.62 now. They also misbehave on 2.60 / 2.61
> 
> Can you try the update .bin above (or wait for 2.6.3). I've increased the stack size.

Not able to get the uploaded .bin.zip to load via the web flasher interface.
Should any switches be set? Full erase and install  or Meshtastic UI ?

resunltd, 2025-03-26:
Log Data at spurious reboot:

Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.

Core 0 register dump:

PC : 0x42051635 PS : 0x00060530 A0 : 0x820330d5 A1 : 0x3fcb14f0

A2 : 0x3fcb152c A3 : 0x3fcb158c A4 : 0xffffffff A5 : 0xffffffff

A6 : 0xffffffff A7 : 0x000000ff A8 : 0x00000041 A9 : 0x3fcb14d0

A10 : 0x00000086 A11 : 0x00000086 A12 : 0x00000003 A13 : 0x3fcb152c

A14 : 0x00000186 A15 : 0x00000000 SAR : 0x0000000c EXCCAUSE: 0x0000001c

EXCVADDR: 0x0000000b LBEG : 0x40056f5c LEND : 0x40056f72 LCOUNT : 0xffffffff

Backtrace: 0x42051632:0x3fcb14f0 0x420330d2:0x3fcb1570 0x4202dec6:0xffffffff |<-CORRUPTED

resunltd, 2025-03-28:
Just now I updated three Indicators to 2.6.3.640e731 from the web flasher.

All have been continuously powered for about 30 minutes. Two from power banks and one from USB 
One unit shows uptime 29:43, battery 
Another, 8:19  with three reboots, battery
Another, 10:22 with three reboots, USB 

Any ideas?
Thanks.

closed 2025-03-28 by thebentern

resunltd, 2025-03-31:
FYI, looking good now.  Over 26 Hours up time with no spurious reboots.

Thank you very much.

and Best Regards.

jpryor223, 2025-04-01:
Same here, good with first 24 hours, no longer reboots.

PR #107 (2025-03-28): fix: add stack protector to cope with array index issues of touch driver
fixes #68 (workaround)
```
0x40377d92: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
0x403816fd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
0x403773da: __stack_chk_fail at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/stack_check.c:35
0x42051d79: LGFXDriver<LGFX_INDICATOR>::touchpad_read(_lv_indev [... cut]