meshtastic/device-ui #82: International Symbols virtual/physical keyboard

Hayvosh, 2025-03-11:
Recently got myself a LilyGo T-Deck Plus and I'm loving it.
However, I might have missed it or if it's missing. 
But I can't find a way to type the characters "Å Ä Ö" that we use in some countries in the EU.
Not via any key combos that I've tried or via the virtual keyboard.

Would be nice if it would be possible to change the keyboard layout on the virtual one, or add the feature to insert those characters somehow.

mverch67, 2025-03-15:
There are 18 languages ... 😅

Hayvosh, 2025-03-15:
> There are 18 languages ... 😅

Then maybe you can enlighten me how to insert them when using virtual keyboard or the physical one when set to english language?

mverch67, 2025-03-15:
> Then maybe you can enlighten me how to insert them when using virtual keyboard or the physical one when set to english language?

In German it's easy:

Ä -> Ae
Ö -> Oe
Ü -> Ue

Hayvosh, 2025-03-15:
> > Then maybe you can enlighten me how to insert them when using virtual keyboard or the physical one when set to english language?
> 
> In German it's easy:
> 
> Ä -> Ae Ö -> Oe Ü -> Ue

I changed from English language to Swedish but there was no difference in the keyboard layout, and if there's any difference I have the white keyboard on my t-deck plus

mverch67, 2025-03-15:
lvgl provides only one standard layout. However, the layout can be customized.

mverch67, 2025-07-30:
> Also there is bulgarian language in interface but it's not working

There's a PR for bulgarian. Can try it and check it: https://github.com/meshtastic/device-ui/pull/148

mverch67, 2026-11-09:
Implemented for physical keyboard: [PR458](https://github.com/meshtastic/device-ui/pull/458)