meshtastic/device-ui #84: Message Interface Improvements

mattdrum, 2025-03-12:
Please add the function of long pressing a message box to copy the contents of the message to the sending box. This would make it much easier to resend messages. 

Additionally, text or symbolic message status from a short press of the message box would be helpful for clarity.

Cloud with an up arrow - Queued on the app to be sent to your device.
Cloud only - Queued on the device to be sent over the mesh.
Cloud with a check mark - At least one other node on the mesh acknowledged the message.
Person with a check mark - The intended recipient of your direct message acknowledged the message.
Cloud crossed out - Not acknowledged or message error.

RealBenAnderson, 2025-03-31:
I agree except I would like to see the last one divided into two icons, one for not acknowledged and another for a key mismatch.

klikva, 2025-05-19:
This would be perfect. I have no specific preferences, every option for resending a message will be very usefull.

dzienisz, 2025-05-22:
I like idea to have two modes Simple and Expert or Communicator and Technical.

I mean one would be strictly for communication purposes and other one for testing the grid and more advanced users. 

The problem right now and with my case was that when I turned it on I didn't know what to do.

A lot of users are not programmers and even not pro users. They just want to enjoy using LoRA network.

juliadin, 2025-09-04:
I am sure it's possible to find ways to communicate the information in a way that is accessible even to non-technical users. Introducing many settings could be a usability problem in itself, in my opinion.

Many of the modern messengers have similar states to worry about and I think the initial suggestion would be ok even for non-technical users if the iconography is explained somewhere. 

Signal for example also uses two check marks... One for network status, one for recipient status. Many people just know that they need to wait for two solid check marks to see if the message was read. Everyone else can discern more information from the symbols, which is nice.

Also from a security perspective: ignoring a key mismatch and not having a way to clearly symbolize it because someone just tries to use it (I don't know if you are referring to this, just guessing) would be a security problem I think. I know about activists and people like that who require short term confidential messaging but have nearly no technical experience. They would greatly benefit from being able to tell that the remote device "not cryptographically sound" anymore to stop messaging it. 

In my opinion hiding this information makes the use of crypto much less meaningful. And even WhatsApp which is for the tech-oblivious masses does it :).

juliadin, 2025-09-04:
Yep, I also agree with that. Let's not forget the original feature request 😅 - that's a very useful feature to have.

kallisti5, 2025-09-22:
An option to "resend" messages which failed to be received would be great too