meshtastic/device-ui #93: Ringtone preview

j6k4m8, 2025-03-18:
**Is your feature request related to a problem? Please describe.**

Right now there's no on-device way (as far as I can tell) to "try out" ringtones.

**Describe the solution you'd like**

It would be nice to have a way to preview ringtones, perhaps when selected, or else to have a menu page for ringtone selection that allows playing separately from selection.

**Describe alternatives you've considered**

I decided to pull up [the source code](https://github.com/meshtastic/device-ui/blob/9ec138c497cbd98c35fac21f2f4e58e0ee1966f1/source/graphics/common/Ringtones.cpp) and try them out individually [online](https://adamonsoon.github.io/rtttl-play/) (thanks @adamonsoon!)

I'm out of my element with this repo but with some guidance from the core devs I'd be happy to take a whack at this.

mverch67, 2025-03-18:
Duplicated to https://github.com/meshtastic/device-ui/issues/48

closed 2025-03-18 by mverch67

j6k4m8, 2025-03-18:
Ah, sorry for the dup! Didn't think to search for "alert" which is the obvious term...