meshtastic/device-ui #102: Support toggling "Mesh via UDP"

vidplace7, 2025-03-23:
Please add support for toggling the new "UDP Multicast" feature added in
- https://github.com/meshtastic/firmware/pull/5779
- https://github.com/meshtastic/protobufs/pull/643

Currently can only be enabled via Python with this config:
```sh
meshtastic ... --set network.enabled_protocols 1
```

Should be a boolean value to enable/disable, I imagine.

vidplace7, 2025-03-23:
For the record, the UDP feature works great on TFT builds :) Would just be nice to be able to toggle it directly from the UI.

mverch67, 2025-03-23:
What's the exact use case? What's the purpose of toggle?

vidplace7, 2025-03-23:
> What's the exact use case? What's the purpose of toggle?

This feature allows the node to mesh with other nodes via UDP Multicast, in addition to LoRa.
The ability to toggle this on the device without having to program it via a Client would be just great!

mverch67, 2025-03-23:
If it is toggled via MUI what data is expected to be seen in a (potentially new?) MUI panel? Just want to understand the use case for the MUI.

vidplace7, 2025-03-23:
> If it is toggled via MUI what data is expected to be seen in a (potentially new?) MUI panel? Just want to understand the use case for the MUI.

Just a toggle in the settings area is all that's expected. A new panel shouldn't be warranted.

UDP meshing simply forwards all packets to UDP Multicast (at a broadcast address) and listens for packets at that same broadcast address. For instance you could have multiple nodes on the same wifi network, each on a different modem preset / frequency, and bridge them via UDP.
MUI devices can participate in this UDP meshing if the config is enabled, and we want that to be easy.

mverch67, 2025-03-23:
I still don't get the exact use case. The reason why I'm asking is because MUI is use case oriented. It's not a tool to toggle arbitrary switches. If there are participants in a UDP network why not show these other nodes as participants? That would make a use case. Otherwise why not have it enabled by default? What is the use case to disable the switch once it is enabled?

tedwardd, 2025-11-14:
To drag this old issue back up. @mverch67 I'm still not sure where the confusion here is. `network.enabled_protocols` is a radio setting like any other firmware setting. If I'm using a standalone device with MUI, I wouldn't expect to need to connect the the device to my phone or laptop to toggle this flag, it should be something I can toggle via the UI on the device. The use case for this feature is pretty clear; to receive mesh packets over UDP multicast on a network connected device. How would I support this feature on a standalone device without a toggle for it in MUI? What am I missing?