meshtastic/device-ui #114: Extended zoom on map tiles

mattdrum, 2025-04-01:
When a zoom level is missing on a map tile and a lower zoom level is available, add the function to step up to the next available higher zoom level and stretch the map tile to cover the lower zoom levels. This would provide sufficient visual reference for the map without needing to have high resolution map tiles. 

Ideally this would extend all the way to the highest zoom level although it would not necessarily be useful it would provide a bit more continuity.

RealBenAnderson, 2025-04-09:
Agreed, scaling the nearest available map tiles up/down is preferable to simply showing nothing.

nwilde1590, 2025-05-15:
Piggybacking to add that using the buttons to move around the map is less convenient on a touchscreen than just dragging the map around. Not sure if there’s a technical reason preventing this (I’d guess maybe something with the speed of rendering updates as you drag the map), but I was curious to bring it up.

dzienisz, 2025-07-23:
hmm... we should definitely have better tutorials on how to make maps or have already made countries and some CDN for it.

mverch67, 2025-07-23:
> hmm... we should definitely have better tutorials on how to make maps or have already made countries and some CDN for it.

what parts of the tutorial you could not follow?
See e.g. a comment from a [novice user](https://github.com/mattdrum/map-tile-downloader/issues/8)!

juliadin, 2025-09-04:
This feature would also be an accessibility win because many of the devices have pretty small displays and maps tend to add more details when zooming in and sometimes even make the things I want to see smaller 😅. I'd like to see this feature :)