meshtastic/device-ui #119: MUI LVGL Colors wrong

Canzel, 2025-04-21:
MUI display inverted or otherwise wrong colors on T-DECK

Expected behavior**
I expected the colors to resemble those on https://meshtastic.org/docs/software/meshtastic-ui/
Greens are red and so on.

**Screenshots / Photos**
Screenshot of map-view with standard OpenStreetMap-tiles, colors all wrong.

**Device (please complete the following information):**
 - Type: T-Deck
 - Exact version: 2.6.4.b89355f Beta 
 - Firmware download source: web-flasher] 
 - Other attached HW: a GPS

Is the setup for LVGL -display library wrong?

[image]

mverch67, 2025-04-21:
Seems your display has some issues. Can you load the Lilygo test software from their repository and check again.

dzienisz, 2025-05-22:
I think in case of colors they are ok (in code) but T-Deck display is not a great display.