meshtastic/device-ui #123: T-HMI / CYD / Mesh-TAB not works...

SQ9PXB, 2025-04-27:
**T-HMI / CYD / Mesh-TAB  requires RX/TX (RS232) connection**

**When compiling T-HMI, CYD ESP32-2432S028RV2 (with PSRAM), MESH-TABB etc**

On screen only display : Waits for TX/RX connection (when connecting GPS or TERM 38800 8N1 it writes "connect"), does nothing else, does not work. 

T-HMI log with GPS on pin 15/16, on screen display "Connected!", and its all... do nothing
MeshEnvelope: no magic found, skipping 6 bytes (008080...)
received 11 bytes from serial
MeshEnvelope: no magic found, skipping 11 bytes (788080...)
received 10 bytes from serial
MeshEnvelope: no magic found, skipping 10 bytes (788078...)
received 13 bytes from serial
MeshEnvelope: no magic found, skipping 13 bytes (780078...)
received 21 bytes from serial
MeshEnvelope: no magic found, skipping 21 bytes (808078...)
received 14 bytes from serial
MeshEnvelope: no magic found, skipping 14 bytes (78f800...)
received 16 bytes from serial
MeshEnvelope: no magic found, skipping 16 bytes (780078...)

mverch67, 2025-04-27:
It definitely works. You probably haven't configured the serial interface of lora device. It must be in mode PROTO.

SQ9PXB, 2025-04-27:
Sorry for my little knowledge on the subject. Can you describe exactly how to set this up? I know a lot of people have trouble with this...
I haven't found a description anywhere. I conclude that due to the small number of available ports, communication is to be via RS232 - is that correct?
Please, write in simple words for a layman, how to configure the system for T-HMI & LORA ex. RA-62... Nothing else. 

Regards,

Matt SQ9PXB

mverch67, 2025-04-27:
You'll have to connect the CYD device to a LoRa device via serial (BT/WIFI may work in future) like Heltec, Rak, or any other device that cannot run MUI natively. So the CYD serves as a MUI display for the connected LoRa device. The serial cable is for the protobuf communication between the two devices. 

[image]

The configuration of the serial port is described in the link to the docs of my previous comment, e.g.
`meshtastic --set serial.enabled true --set serial.baud 38400 --set serial.mode PROTO --set serial.rxd 8 --set serial.txd 6`

Note, that the rx/tx ports depends on the LoRa device and also that _meshtastic CLI_ recently had problems sending these commands successfully. Android or Web-Client should work though.

SQ9PXB, 2025-04-27:
Thank you. Now everything is clear. I will test it soon.

Btw, I thought T-HMI would natively support MUI....

Regards

mverch67, 2025-04-27:
> Btw, I thought T-HMI would natively support MUI....

It does as you can see in the picture. If you can find 6 free pins you could also attach a LoRa chip and build and run meshtastic firmware on it.

closed 2025-04-28 by mverch67

SQ9PXB, 2025-05-03:
Hello,

Sorry for the long delay, but work duties took up time.

I confirm, it works as you described. I connected T-HMI to the ESP32-S3 module via RS232.

In T-HMI you can get free GPIO at the cost of an SD card (I know, there will be no maps)

Regards
SQ9PXB