meshtastic/device-ui #134: Programming mode screen shows the "fixedPin" value, even when in "RANDOM_PIN" mode

moonbench, 2025-05-24:
**Describe the bug**
When rebooting into the bluetooth programming mode, where it shows a screen with text like:
```
>> Programming mode <<
        123456
```
it appears that the displayed PIN is always the `bluetooth.fixedPin` value, even if the device is in `RANDOM_PIN` mode.

I encountered this when trying to pair with a T-Deck Plus for the first time over bluetooth and the pairing operation repeatedly failed when using the displayed value of `123456`. I was suspicious that the shown PIN was incorrect because entering `meshtastic --info` in the terminal showed that my device should be in `RANDOM_PIN` mode. After switching to `FIXED_PIN` mode I was able to pair with to the device. Then, even after switching back to `RANDOM_PIN` mode, it continued to display the fixed PIN instead of a new random one.

As a consequence, while the device is in `RANDOM_PIN` mode there's effectively no way to pair with it because the randomly generated PIN isn't shown.

**To Reproduce**
Steps to reproduce the initial behavior:
1. Start from a fresh installation/configuration
2. Reboot the device into bluetooth programming mode
3. It should display the default fixed PIN of `123456` instead of a random one
4. Attempting to pair with the device should fail using that PIN
  1. Since the device defaults to `RANDOM_PIN` mode and we're not seeing the real random PIN 

Steps to indicate that it's showing the `bluetooth.fixedPin` value:
1. Using the terminal, change the device to `FIXED_PIN` mode
  1. `meshtastic --set bluetooth.mode FIXED_PIN`
2. Change the PIN to a new value
  1. `meshtastic --set bluetooth.fixedPin 555555`
3. The device should reboot and show the new PIN (and allow pairing)
4. Change the device back to `RANDOM_PIN` mode
  1. `meshtastic --set bluetooth.mode RANDOM_PIN`
5. The device should reboot, but it will continue to display `555555` (but no longer allow pairing with that PIN)

**Expected behavior**
The programming mode screen should show the random pin when in `RANDOM_PIN` mode, instead of the fixed PIN.

**Screenshots / Photos**
_I can attach photos if needed. But it's just the programming mode screen with the default pin shown._

**Device (please complete the following information):**
 - Type: T-Deck Plus
 - Exact version: 2.6.4
 - Firmware download source: Web-flasher
 - Other attached HW: Stock T-Deck Plus

**Additional logs**
While displaying `123456` on the screen, despite not being in `FIXED_PIN` mode, this was my device's bluetooth configuration:
```
  "bluetooth": {
    "enabled": true,
    "fixedPin": 123456,
    "mode": "RANDOM_PIN"
  },
```

mverch67, 2025-05-24:
When entering the programming mode MUI has to 
a) disable wifi
b) enable bluetooth
c) set the fixedPin mode 
d) set a random fixedPin value

After the reboot this fixed random Pin is shown on the boot screen of the MUI display (whether local or remote).

The user is not supposed to change any of the bluetooth settings which interferes with these required settings and which will result in (obviously) unexpected behavior.

moonbench, 2025-05-24:
To clarify, I only changed the settings after I noticed the unexpected behavior.

Starting from an fresh installation, the PIN shown the programming mode screen on my device is `123456` but that PIN didn't work for pairing.

PR #135 (2025-05-24): fix: set bluetooth fixedPin to random value (restart menu)
fixes #134
closed 2025-05-24 by mverch67