meshtastic/device-ui #152: Difficulty setting T-Deck Brightness to 100%

aseiger, 2025-06-17:
**Describe the bug**
Dragging the T-Deck slider to 100% brightness results in the display not going to 100%... sometimes... despite the slider reading 100%.

Re-booting the T-Deck with the display at 100% almost always results in the display coming back up at 50% or so.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to adjust screen brightness
2. Drag slider all the way to the right
3. Note how display brightness seems to go to 100% briefly, then suddenly back to 50% or so
4. Get screen brightness to 100%, then power-cycle, observe screen brightness at 50% or so.

**Expected Behavior**
I'd expect that the display brightness would go to 100% when the slider is pulled to 100%, and have that setting persist across reboots.

**Device (please complete the following information):**
 - Type: T-Deck
 - Exact version 2.6.10.9cce4455
 - Firmware download source Web Flasher
 - Other attached HW: Unknown GPS Module

mverch67, 2025-06-17:
This seems to be a hardware issue of the T-Deck. It doesn’t happen on any other device.
So nothing we can do about.

aseiger, 2025-06-17:
Weird - is the t-deck using weird PWM or something? Your fix seems like it would work, I have found that 99% brightness does not have these issues. As long as it actually says 100% so my brain can be quiet. 

I need to find more time - I do LVGL all the time as my day job, I'd love to help y'all out.

PR #154 (2025-06-20): fix: workaround: T-Deck 100% brightness issue
closes #152