meshtastic/device-ui #178: T-Deck Plus: Shortcuts not working

ftrueck, 2025-08-15:
**Describe the bug**
None of the FN-Type Shortcuts work on my device. The FN-Symbol is never shown on screen as the FN-Overlay is not activated by Alt + C.

**To Reproduce**
Try Alt + C on newly flashed T-Deck Plus.

**Expected behavior**
FN-Layer is shown.

**Device (please complete the following information):**
 - Type: T-Deck Plus
 - v2.6.11, v2.7.5
 - Firmware download source: web flasher
 - original T-Deck Plus

**Additional logs**

DEBUG | 17:33:38 97 [DeviceUI] key press value: 99    <- C-Key only
DEBUG | 17:33:38 97 [DeviceUI] key press value: 99
DEBUG | 17:33:38 97 [DeviceUI] key press value: 99
DEBUG | 17:33:38 97 [DeviceUI] key press value: 99
DEBUG | 17:33:39 98 [DeviceUI] key press value: 12     <- ALT + C
DEBUG | 17:33:40 99 [DeviceUI] key press value: 103   <- G- Key

ftrueck, 2025-08-15:
So I should post this issue in firmware instead?

ftrueck, 2025-08-16:
ALT + G does not work either, by the way.

mverch67, 2025-08-16:
It's a different UI. What you are referring to is BaseUI. If you goto Settings->Reboot/Shutdown, long press the BT button and confirm switching to BaseUI it will work.

ftrueck, 2025-08-16:
> It's a different UI. What you are referring to is BaseUI. If you goto Settings->Reboot/Shutdown, long press the BT button and confirm switching to BaseUI it will work.

If I long press the BT button it just reboots to programming mode. I can not select any UI. I just see the standard MUI that is installed with flashing the firmware.

ftrueck, 2025-08-16:
I see. I've flashed 2.7.5 and I can confirm in Base UI the FN-Shortcut for GPS works. But what about MUI?

mverch67, 2025-08-16:
MUI does not need it because unless the OLED style UI all these functions can be accessed via settings menu and home screen shortcuts.

ftrueck, 2025-08-16:
Not sure if I can consent. If I long press the GPS button in MUI the device reboots (rebooting message). If I enable/disable gps in base UI this does not happen. A reboot does only make sense if the GPS feature gets disabled completely (not present) but not to switch off or on GPS usage. Single tapping on the GPS button does nothing also no debug output when tapping the icon.

mverch67, 2025-08-16:
Enable/disable GPS without reboot is not supported via the API. The shortcut is a firmware internal function that does not use the API.

mverch67, 2025-08-19:
> Please do not feel offended, but why does the API require a reboot for disable / enable the GPS? If I understand correctly the API is part of the firmware. So it should also have access to the internal features that the firmware has. So I do not understand why it is necessary to reset.

I think in general this is due to embedded arduino libraries saving every extra byte to squeeze the code into a 20kB Arduino.
So one function that is missing all over the place is the delete/destructor cleanup, cause it's not needed because the device has a reset button.
For GPS in particular this disable function has been added later using double click at the device. So far there was no reason to amend the API (the phone is using the phone GPS), but updating the API for GPS would allow some more feature for MUI (such as heading, real-time tracking, etc.)

dzienisz, 2025-08-19:
Where I can find all T-Deck shortcuts and Meshtastic UI shortcuts?

mverch67, 2025-08-27:
Will be followed up by https://github.com/meshtastic/firmware/issues/7687