meshtastic/device-ui #204: Demiss Message delete all Messages bot only the current one

Delta1977, 2025-09-15:
**Describe the bug**
Demiss Message delete all Messages bot only the current one 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Messages'
2. Click long
3. Select demiss
4. all Messages are deleted

**Expected behavior**
only the current message should be deleted

**Screenshots / Photos**
If applicable, add screenshots or photos to help explain your problem.

**Device (please complete the following information):**
 - Type: [Heltec v3]
 - Exact version [2.7.7]
 - Web flash

mverch67, 2025-09-16:
The expected behavior is to delete the entire chat. Besides that it cannot work with a heltec V3.

mverch67, 2025-09-24:
> Why is cannot work with a heltec v3 ? Why not make a option to delete "one by one"

Maybe you are talking about a different UI? Android or iOS?